kennetek/gridfinity-rebuilt-openscad #194: Printing errors and artifacts

Pytonballoon810, 2024-06-06:
I created this bin using the minimal implementation:
[image]
and printed it twice and both look like this:
[image]
[image]
In the slicer it looks like this:
[image]
the "closing layer" is also just this one top layer.

EmperorArthur, 2024-06-15:
Waiting for my spiral vase fix to be merged before I start down this rabbit hole.

ch3vr0n5, 2024-06-18:
This is interesting. I fixed this in a pull request months ago. #126 & #137 

I haven't pulled the latest changes since then, so my container looks like this:

[image]

I'll have to pull the changes and see if this is a regression.

EmperorArthur, 2024-06-27:
> I'll have to pull the changes and see if this is a regression.

I completely re-did wall generation based on the Gridfinity spec, and I know it caused issues with Spiral vase mode.  #181

Depending on how the cutter for the lite bins is done, there may be some incompatibilities.  I do have test cases in my branches, and can add more for this as well.

closed 2024-11-06 by EmperorArthur