kennetek/gridfinity-rebuilt-openscad #195: Feature Request: An option for the Gridfinity Refined central thumbscrew hole

TechyShishy, 2024-06-19:
The stl of the screw hole in case you want to import() it:

https://files.printables.com/media/prints/413761/stls/3590032_b22e71f5-c9dc-4873-9745-da7f4e5e899a/bin-screw-hole.stl

The author describes it as such:

> The screw hole in the bin is an ISO Metric Profile, 15.0mm size, M15x1.5 designation, 6H class hole. There is an offset to the thread faces of 0.125mm to provide a little bit of slop to make it easier to secure the thumbscrew.

https://www.printables.com/model/413761-gridfinity-refined

EmperorArthur, 2024-06-20:
Hmm, that linked to https://www.printables.com/model/417152-gridfinity-specification  Now to compare that to the PRs which I already have in the pipe to generate things based on the spec.

It's an interesting idea, but I'm going to wait until at least PR #179 is merged to start work.

andrewguy, 2024-10-03:
Hi @TechyShishy, see the above pull request if you want these screw holes as an option. I didn't use the provided STL - it didn't play well with OpenSCAD when performing a `difference()` operation. I've used the [threads-scad library ](https://github.com/rcolyer/threads-scad) instead. I didn't include the 0.125mm offset as it didn't seem necessary based on my test prints (unless it's somehow included in the threads-scad library). Hope it works for you!