kennetek/gridfinity-rebuilt-openscad #202: Skeletonized styles are not working anymore

freiser77, 2024-07-10:
Hello,

I want to generate a 4x3 baseplate in the skeletonized style with magnet holes and screw together possibility.
[image]

But the skeleton does not work. The inside is stilled filled and so it is not possible to screw the baseplates together.
[image]

If I disable the magnet hole and the vertical screw holes, it works.
Is there a bug?

Greetings

Florian

EmperorArthur, 2024-07-13:
Fixed by #187, which has been awaiting merge since May 17th.

jhoblitt, 2024-08-28:
As #187 was merged, can this issue be closed?

closed 2024-08-28 by freiser77