kennetek/gridfinity-rebuilt-openscad #206: Only Corners is broken after #179

dudu631, 2024-07-31:
After the merge of [PR179](https://github.com/kennetek/gridfinity-rebuilt-openscad/pull/179), only_corners options is throwing errors:

```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
[ERROR: Assertion 'is_list(hole_options)' failed in file gridfinity-rebuilt-holes.scad, line 247](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/247,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/gridfinity-rebuilt-holes.scad)
[TRACE: called by 'assert' in file gridfinity-rebuilt-holes.scad, line 247](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/247,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/gridfinity-rebuilt-holes.scad)
[TRACE: call of 'block_base_hole(hole_options = 0, o = 0)' in file gridfinity-rebuilt-holes.scad, line 246](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/246,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/gridfinity-rebuilt-holes.scad)
[TRACE: called by 'block_base_hole' in file gridfinity-rebuilt-utility.scad, line 271](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/271,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/gridfinity-rebuilt-utility.scad)
[TRACE: called by 'translate' in file gridfinity-rebuilt-utility.scad, line 270](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/270,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/gridfinity-rebuilt-utility.scad)
[TRACE: called by 'children' in file generic-helpers.scad, line 40](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/40,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/generic-helpers.scad)
[TRACE: called by 'rotate' in file generic-helpers.scad, line 39](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/39,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/generic-helpers.scad)
[TRACE: called by 'for' in file generic-helpers.scad, line 38](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/38,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/generic-helpers.scad)
[TRACE: call of 'pattern_circular(n = 4)' in file generic-helpers.scad, line 37](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/37,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/generic-helpers.scad)
[TRACE: called by 'pattern_circular' in file gridfinity-rebuilt-utility.scad, line 269](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/269,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/gridfinity-rebuilt-utility.scad)
[TRACE: called by 'difference' in file gridfinity-rebuilt-utility.scad, line 266](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/266,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/gridfinity-rebuilt-utility.scad)
[TRACE: called by 'render' in file gridfinity-rebuilt-utility.scad, line 265](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/265,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/gridfinity-rebuilt-utility.scad)
[TRACE: call of 'block_base(gx = 3, gy = 3, l = 42, dbnx = 1, dbny = 1, hole_options = 0, off = 0)' in file gridfinity-rebuilt-utility.scad, line 264](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/264,C:/Users/Eduardo/Downloads/gridfinity-rebuilt-openscad-main%20(1)/gridfinity-rebuilt-openscad-main/gridfinity-rebuilt-utility.scad)
Compiling design (CSG Products generation)...
Geometries in cache: 421
Geometry cache size in bytes: 3817632
CGAL Polyhedrons in cache: 64
CGAL cache size in bytes: 2095336
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.037
```

EmperorArthur, 2024-08-05:
I will ensure there are tests for this issue, and examine the current behavior, and if it is fixed in one of the open Pull Requests.

EmperorArthur, 2024-08-09:
Confirmed in master.  I will add a test, and create a PR to fix it.

EmperorArthur, 2024-09-01:
Fixed by #213

closed 2024-09-01 by EmperorArthur