kennetek/gridfinity-rebuilt-openscad #210: Basic Tutorial on how to Open the Program

Doubs115, 2024-08-12:
Something I had an issue with when using this program was that the info required was scattered and not straightforward. 

The website is good, but if one happens to land on the site before the GitHub page, there doesn't seem to be a link saying “GO HERE”.
If there is one, then ideally it should be more obvious. 

2nd, as someone who has never touched openSCAD before, both the download and getting it to function need at least a basic tutorial.
I admit I might not be the smartest tool in the shed but considering I got it to work I would say I am at least competent. 

I thought, like other programs, that I would only need the certain scad file to run a certain configuration. I then loaded it into OpenSCAD only to get a bunch of errors. (In hindsight, duh, I was missing 90% of the program.)
So ideally it should be stated clearly to download the whole GitHub repo and then to open scad files from the extracted zip folder.

Lastly, not that it's the fault of this program, but downloading openSCAD wasn't strait forward either. So, a mini tutorial on getting the beta for the fast-cfg feature would help reduce friction. Even if it was just a couple images showing the most critical points in a step by step would be better than nothing. 

So take these with some salt, but as a new user, these are what I struggled with the most and almost made me give up on trying to figure it out. Otherwise, really cool program, and it's going to save me a ton of time.

EmperorArthur, 2024-09-01:
Good suggestions.  I've been doing the classic of documenting in code and ignoring the site documentation.

Getting the latest version of OpenSCAD is something even we struggle with.  Especially for CI/CD.  There is no "beta" of OpenSCAD.  Rather there's only nightly releases.  Which means we don't have any guarantees of stability!  I would suggest creating/commenting on an issue over there for a beta release.

I am also considering moving some of the utility functions to the "src" folder, and can add a note to the README that you must download the entire repo, along with basic instructions on how to do so via zip file or git.

wcndave, 2024-10-10:
I second this, however here's what I found so far.

I already had openscad, and downloading/installing was not a problem.
There is not fast-csg in latest nightly version, so assume it's just built in now.

It was not clear which of the scad files one should open for which purpose...
I guess -baseplace and -bins, however what's vase for? and what's -lite for?

Also the actual parameters on the docs are not always clear, like what are refined holes? compared to magnet holes.
I did find I needed a lot of "trying changing things and see what happens"

But given it's for free it's pretty amazing, so can't really complain!  Good job!

EmperorArthur, 2024-10-13:
Thanks for the feedback!  I really appreciate it, since it's so easy to just "know" things after so long.  I also spend most of my time documenting back-end things, that I forget to do the front-end.

Some positives at least.
* PR #229 removes options which even confused me.  Especially since the code is non-trivial to understand.
* PR #215 fixed some incorrect documentation relating to height.
* While I'm far from done, I've tried to add assertions to prevent bad options, or things that are not supposed to be allowed from occurring.

On the 3nd note, @wcndave are there any parameters / values you've found that cause something wacky or unexpected?

I am also debating changing a few things so the output window will have things like bin size information.  Thoughts?

reedy, 2024-10-15:
> There is not fast-csg in latest nightly version, so assume it's just built in now.

https://www.reddit.com/r/openscad/comments/16rcdiu/cant_find_fastcsg/

> fast-csg is already obsolete. The manifold rendering engine is super fast, even faster than fast-csg. To use it you need a dev snapshot. Then enable manifold on the feature tab.

`manifold` seems to already be obsolete too...

I guess the various comments in the repo should be updated https://github.com/search?q=repo%3Akennetek%2Fgridfinity-rebuilt-openscad%20%20%20fast-csg&type=code

Do we want to remove `'--enable=fast-csg',` from the test runner too?

reedy, 2024-10-15:
Would a simple TLDR type tutorial in the main README make sense?

1) Download OpenSCAD (probably a dev snapshot)
2) Git clone/grab a zip of the repo
3) Open the files (different docs labelling what each base file is) in OpenSCAD
4) ???
5) Profit!