kennetek/gridfinity-rebuilt-openscad #216: Symetrical Bins

0zerotalent, 2024-09-10:
I'm wondering how hard it would be to create symmetrical bins when you have a non-integer number of bases.
ie currently if you set 2.4 bases along the x-axis it will print 2 bases with 0.4 added to one end. I would like it to print 0.2 bases at both ends. This way I can place my full grid in the middle of a draw and have a bin that overhangs it on both sides.

EmperorArthur, 2024-10-13:
This is tricky for multiple reasons.

PR #229 would limit bin sizes to multiples of 0.25, so you would have to go for 2.5 instead of 2.4, and it would automatically make all bins fit on half bases.  That may do part of what you are asking for, since it does make bins symmetrical along that axis.

Do you think both bases should be on 1/2 divisions if either base is?

# Images (2.5 x 2)

PR #229:
[image]

Alternative:
[image]

EmperorArthur, 2024-10-13:
I can think of a way to code what you want using gridfinityInit, and gridfinityBase with custom options, but please be patient as I can't get to it right now.

phrxmd, 2024-12-22:
> Do you think both bases should be on 1/2 divisions if either base is?

I'm not the OP, but I'm one of the people playing around with fractional grids. Square fractional bases are always preferable, so that you can rotate the bin 90° and it will still fit the grid.