kennetek/gridfinity-rebuilt-openscad #217: Not creating scoop

jottr, 2024-09-11:
I can't get the scoop feature working. 
I'm working on a clone of most recent `main` branch.

[image]

```openscad
/* [General Settings] */
// number of bases along x-axis
gridx = 1; //.5
// number of bases along y-axis
gridy = 1; //.5
// bin height. See bin height information and "gridz_define" below.
gridz = 6; //.1

/* [Linear Compartments] */
// number of X Divisions (set to zero to have solid bin)
divx = 0;
// number of Y Divisions (set to zero to have solid bin)
divy = 0;

/* [Cylindrical Compartments] */
// number of cylindrical X Divisions (mutually exclusive to Linear Compartments)
cdivx = 0;
// number of cylindrical Y Divisions (mutually exclusive to Linear Compartments)
cdivy = 0;
// orientation
c_orientation = 2; // [0: x direction, 1: y direction, 2: z direction]
// diameter of cylindrical cut outs
cd = 10; // .1
// cylinder height
ch = 1;  //.1
// spacing to lid
c_depth = 1;
// chamfer around the top rim of the holes
c_chamfer = 0.5; // .1

/* [Height] */
// determine what the variable "gridz" applies to based on your use case
gridz_define = 0; // [0:gridz is the height of bins in units of 7mm increments - Zack's method,1:gridz is the internal height in millimeters, 2:gridz is the overall external height of the bin in millimeters]
// overrides internal block height of bin (for solid containers). Leave zero for default height. Units: mm
height_internal = 1;
// snap gridz height to nearest 7mm increment
enable_zsnap = false;

/* [Features] */
// the type of tabs
style_tab = 0; //[0:Full,1:Auto,2:Left,3:Center,4:Right,5:None]
// how should the top lip act
style_lip = 0; //[0: Regular lip, 1:remove lip subtractively, 2: remove lip and retain height]
// scoop weight percentage. 0 disables scoop, 1 is regular scoop. Any real number will scale the scoop.
scoop = 1; //[0:0.1:1]

/* [Base] */
// number of divisions per 1 unit of base along the X axis. (default 1, only use integers. 0 means automatically guess the right division)
div_base_x = 0;
// number of divisions per 1 unit of base along the Y axis. (default 1, only use integers. 0 means automatically guess the right division)
div_base_y = 0;

/* [Base Hole Options] */
// only cut magnet/screw holes at the corners of the bin to save uneccesary print time
only_corners = false;
//Use gridfinity refined hole style. Not compatible with magnet_holes!
refined_holes = true;
// Base will have holes for 6mm Diameter x 2mm high magnets.
magnet_holes = false;
// Base will have holes for M3 screws.
screw_holes = false;
// Magnet holes will have crush ribs to hold the magnet.
crush_ribs = true;
// Magnet/Screw holes will have a chamfer to ease insertion.
chamfer_holes = true;
// Magnet/Screw holes will be printed so supports are not needed.
printable_hole_top = true;
```

GiovanH, 2024-09-13:
How are you calling the function?

jottr, 2024-09-13:
What do you mean, how do I call it? I  render the `gridfinity-rebuilt-bins.scad` from within openSCAD.
 Here's the full patch:

```git 

diff --git a/gridfinity-rebuilt-bins.scad b/gridfinity-rebuilt-bins.scad
index 3b36dba..b352846 100644
--- a/gridfinity-rebuilt-bins.scad
+++ b/gridfinity-rebuilt-bins.scad
@@ -30,9 +30,9 @@ $fs = 0.25; // .01
 
 /* [General Settings] */
 // number of bases along x-axis
-gridx = 3; //.5
+gridx = 1; //.5
 // number of bases along y-axis
-gridy = 2; //.5
+gridy = 1; //.5
 // bin height. See bin height information and "gridz_define" below.
 gridz = 6; //.1
 
@@ -62,13 +62,13 @@ c_chamfer = 0.5; // .1
 // determine what the variable "gridz" applies to based on your use case
 gridz_define = 0; // [0:gridz is the height of bins in units of 7mm increments - Zack's method,1:gridz is the internal height in millimeters, 2:gridz is the overall external height of the bin in millimeters]
 // overrides internal block height of bin (for solid containers). Leave zero for default height. Units: mm
-height_internal = 0;
+height_internal = 1;
 // snap gridz height to nearest 7mm increment
 enable_zsnap = false;
 
 /* [Features] */
 // the type of tabs
-style_tab = 1; //[0:Full,1:Auto,2:Left,3:Center,4:Right,5:None]
+style_tab = 0; //[0:Full,1:Auto,2:Left,3:Center,4:Right,5:None]
 // how should the top lip act
 style_lip = 0; //[0: Regular lip, 1:remove lip subtractively, 2: remove lip and retain height]
 // scoop weight percentage. 0 disables scoop, 1 is regular scoop. Any real number will scale the scoop.

```

kennetek, 2024-09-13:
Change `height_internal` back to zero. Only need to mess with that parameter if you want solid bins.

jottr, 2024-09-13:
It's exactly the opposite for me:

`height_internal=0`:

[image]

`height_internal=1`:

[image]

kennetek, 2024-09-13:
Change both `divx` and `divy` to `1`. I don't know why its zero by default.

jottr, 2024-09-14:
Ok, that fixed it.

jottr, 2024-09-14:
@kennetek may I suggest to set the defaults of `divx` and `divy` to `1`?

closed 2024-09-14 by jottr