kennetek/gridfinity-rebuilt-openscad #224: Empty layers between bottom and bin

tobymurray, 2024-09-23:
Using:
- gridfinity-rebuilt-lite.scad @ 09e8cf0
- [OpenSCAD](https://www.openscad.org/) version 2024.09.15.ai20377 (git df9595dea)

I'm getting empty layers:
[image]

Rolling back to 51f1198 seems to resolve it, so I'm guessing either I'm doing something wrong or there's a regression in https://github.com/kennetek/gridfinity-rebuilt-openscad/pull/214?

EmperorArthur, 2024-10-04:
Regarding the render time, which version of OpenSCAD are you using, and do you have fastCSG turned on?

I'll take a look, and possibly add some test cases.

tobymurray, 2024-10-04:
Using:

- gridfinity-rebuilt-lite.scad @ https://github.com/kennetek/gridfinity-rebuilt-openscad/commit/09e8cf0de9c0383cc61275f10fcca3cc5b9ba405
- [OpenSCAD](https://www.openscad.org/) version 2024.09.15.ai20377 (git df9595dea)

I am not using fastCSG

EmperorArthur, 2024-10-06:
Thanks.  I see the same render time issue, and am working the real problem now.

EmperorArthur, 2024-10-13:
I've identified the problem, and am working on a fix.  I actually had most of one done, but it wasn't compatible with some of the latest PRs, and I prefer to not destroy other people's hard work.

# Technical Details
For reference, the issue is how offset (off) works.
Lite is **currently** expecting a positive offset to shrink holes in [x, y, z] dimensions, and shrink walls in the [x, y] dimensions.  Holes work as expected, but walls do not. 

I will be making the following changes:

* Adjusting offset so a positive value grows holes, and a negative value shrinks them.
* Creating a "gridfinityBaseLite" module which handles things internally.
* This module will also handle gluing the bases together, allowing for a purely lite base without a top.

PR #243 (2024-11-06): Base Updates. Fixes lite bins.
Re-worked the base creation code. Again...

Lite bins now look like this (3x3 base, 2x2 compartments):
[image]

* Fixes #242
* Fixes #232
* Fixes #224
* May Fix #212
* Fixes # #194

The lite bases don't have quite as many options as the regular bins.  However for the simple case this should be a nice solution.  It also massively simplifies the code.

I am thinking of moving base generation code to a separate file, but this is good enough for the moment.
closed 2024-11-06 by EmperorArthur