kennetek/gridfinity-rebuilt-openscad #225: Weird Cutouts in bin (using the lite-rebuild)

Pytonballoon810, 2024-09-30:
Had this issue on the newest version of the rebuild. All settings included in the screenshots

These are the settings i put in:
[image]
[image]
After rendering it looks like this:
[image]
[image]
In the slicer it looks like this:
[image]

tobymurray, 2024-10-04:
Looks like this is the same as https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/224

EmperorArthur, 2024-10-06:
I see the issue, and am working on fixing it.

EmperorArthur, 2024-10-13:
Closing as a duplicate of #224.

I am actively working the issue.

closed 2024-10-13 by EmperorArthur