kennetek/gridfinity-rebuilt-openscad #235: Latest Main broken

feliperiffel, 2024-10-15:
In the latest 2 commit the generation of bins is not working as expected, parameter updated in the left pannel are not being applied properly and some issues during the render with some artifacts, I reverted to the latest commit that the test were passing and it's working.

EmperorArthur, 2024-10-17:
Please provide images of the issue, and which exact configuration settings were not working.

Additionally, please provide what version of OpenSCAD you are using.

feliperiffel, 2024-10-17:
Sorry, when I pull the last main again to get the images, all start to work fine, I do not know if was some mess in git, or dirty between branches change, but it's working as expected, closing the issue now.