kennetek/gridfinity-rebuilt-openscad #249: Gridfinity-rebuilt-lite issue

mojamuto, 2024-11-17:
Hi, when I create a cube with gridfinity lite I can't stack one with another, and if I put a cube created with lite and another normal cube they are not the same.

[image]
the cube on the left is made with gridfinity-lite the one on the right with bins.

When passing it through orcaslicer the bridges are also made at different heights:
[image]
the cube made with gridfinity lite makes the bridge 5 layers before the one built with gridfinity-bins (1mm difference)

my version of openscad is:
[image]

EmperorArthur, 2024-11-17:
Thanks for the report.  
The bridging appears to be a separate issue, and I'd appreciate it if you would create a new issue just for that.

The bin on the left looks like it has something clipping into the base.  There should not be a flat overhang like you are seeing.  Does it occur on a 2x2 or 3x3 bin as well?

mojamuto, 2024-11-17:
Yes, it occurs in all sizes.

[image]

the left one is 2x2 and the right one is 3x3.

I will open another problem just for the bridges, thank you very much.

wjugit, 2024-11-27:
In case this helps....
I think if you just modify the following **translate** statement in **gridfinity_base_lite** (src/core/gridfinity-rebuilt-utility.scad)  the lite bins will no longer have that overhang.   

    //Bridging structure to tie the bases together
    difference() {
  // translate([0, 0, BASE_HEIGHT-top_bottom_thickness])
      **translate([0, 0, BASE_HEIGHT-top_bottom_thickness/2])**

wjugit, 2024-11-27:
Note:  **translate([0, 0, BASE_HEIGHT])** produces an even cleaner exterior but not sure if there are any downsides across the interior of the bins

Pytonballoon810, 2025-03-23:
I have fixed this issue.

cporterbassett, 2025-07-02:
I tested to see if @EmperorArthur 's PR fixed this issue.

By my testing, it does not.

Using the file `gridfinity-rebuilt-lite.scad` in the root of the repo, I still get those same overhangs.  

Here is a screen shot:

[image]

cporterbassett, 2025-07-02:
On the other hand, the [commit](https://github.com/Pytonballoon810/gridfinity-rebuilt-openscad/commit/db2ad54f8804b1b2e364bf23c59e1289df327578) by @Pytonballoon810 linked above worked for me.

closed 2025-08-31 by EmperorArthur