kennetek/gridfinity-rebuilt-openscad #252: Thin Style Baseplates Not Rendering Properly

buildxyz-git, 2024-11-21:
Excellent project!

I wanted to make a simple thin frame but even just setting this with no other option set gives me odd rendering results:
```
// baseplate styles
style_plate = 0; // [0: thin, 1:weighted, 2:skeletonized, 3: screw together, 4: screw together minimal]
```

for example, a 2x2 grid previews as:
[image]

but renders as:
[image]

Am I messing something up here?

wjugit, 2024-11-22:
I ran into a similar issue.  Resolved it locally by _uncommenting_ the final [0,0] entry in the definition of BASEPLATE_LIP in src/core/standard.scad (see below).   This may also fix #246 and #247.   

BASEPLATE_LIP = [
    [0, 0], // Innermost bottom point
    [0.7, 0.7], // Up and out at a 45 degree angle
    [0.7, (0.7+1.8)], // Straight up
    [(0.7+2.15), (0.7+1.8+2.15)], // Up and out at a 45 degree angle
    [(0.7+2.15), 0], // Straight down
    **[0, 0] // Implicit back to start**
];

buildxyz-git, 2024-11-22:
Yep, that fixed it thank you!

madhephaestus, 2024-11-25:
I made the suggested change in a PR, I hope its merged soon :) I just added this project to [CaDoodle](https://github.com/CommonWealthRobotics/CaDoodle/blob/main/README.md) as a shape generator.

PR #253 (2024-11-27): Fix #252
close https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/252
close #246
commit e1e5dcc4 (2024-11-27): Merge pull request #253 from madhephaestus/patch-1

Fix #252

close #252
close #246
closed 2024-11-27 by EmperorArthur