kennetek/gridfinity-rebuilt-openscad #255: bug: lite bins - style_tab for compartments is broken - single tab is always left

derFliegendeHamburger, 2024-11-26:
with current code (@ 285 commits)  for lite bins the style_tab for bins with compartments is broken - if any, then only onesingle tab is crated, always positioned at top left.

This same feature works well with the code version on printables (but that is too old to contain the new easy way for lite bins to define bottom height)

derFliegendeHamburger, 2024-11-26:
possilbe cause:
for style_tab=0; (all) in module cutEqual() the condition
        if (
            place_tab == 1 && (i != 1 || j != n_divy) // Top-Left Division
        ) {
is triggered for all but the top single compartment and thus calling cut() with t=5; thus preventing creation of a tab.

As I do not understand the intention of this conditional, I cannot offer a viable solution.
My primitve workaround is to replace invocation of cut(5) with cut(style_tab)

derFliegendeHamburger, 2024-11-26:
found likely culprit:
with Commit b4ff706 a new param place_tab was introduced to cutEqual(). Comment describes this:
'// place_tab:   tab suppression for all compartments. see "gridfinity-rebuilt-bins.scad"'
But it was never implemented to set this new param place_tab to 0  when invoking from gridfinity-rebuilt-lite.
Cheers

k4pu77, 2025-01-14:
@derFliegendeHamburger you can fix it by adding the place_tab paramter
```
    cutEqual(n_divx = divx, n_divy = divy, style_tab = style_tab, scoop_weight = 0, place_tab = 0);
```

derFliegendeHamburger, 2025-01-15:
ok, I have it working now. Thanks

closed 2025-01-15 by derFliegendeHamburger