kennetek/gridfinity-rebuilt-openscad #260: Feature Request: Add versioning and cut official releases

leviable, 2024-12-14:
As it says on the tin, it would be useful to start versioning the code and cut official releases.

Example use case on why this would be helpful:
In April, 2024, MakerWorld [added their Parametric Model Maker to their platform](https://www.reddit.com/r/BambuLab/comments/1c1d1x7/embracing_openscad_community_introducing_the/). At that time, or perhaps sometime after, they added support for the `gridfinity-rebuilt-openscad` library within the PMM. Unfortunately, it isn't currently known which commit they used when they incorporated the library, and as this project has received a lot of updates since then it is difficult to know how to interact with the library without significant trial and error.

I would like to prepose the following:

- Beginning versioning the code, using perhaps [semantic versioning](https://semver.org/) or [calendar versioning](https://calver.org/)
- Begin cutting releases to coincide with versioning
- Include a `version` function that reports the current version of the code, and/or the current git hash:
```
_gridfinity_rebuilt_version = 
	[2024, 12, 14];
function strings_version() =
	_gridfinity_reuilbt_version;
```

phrxmd, 2024-12-22:
This would be really useful, I've run into unclear versioning issues a few times.

jpypi, 2025-01-23:
I, too, would love to see this, having built a site in February 2024 (https://gridfinity.perplexinglabs.com) with similar functionality that uses this. Coordinating changes is tricky. I'd suggest semantic versioning, since some changes are just upgrades, some are bug fixes, some are breaking changes. Knowing how the primary input variables are going to change would be incredibly helpful.

EmperorArthur, 2025-01-26:
Hello,

I have no problems adding this feature, but do not have the time to figure out what version was used by MakerWorld. From an API standpoint, Version 1 would have been sometime last year, and Version 2 is in pre-alpha, with many changes made through October/November of last year.

Many functions still make use of hidden global variables, and there currently is no documentation or policy about which functions are meant to be API stable and which are not.

To implement semantic versioning we need the following:

* What functions are being used, and should be stabilized as a minimum API.
* What commit should be tagged as Version 1.0

closed 2025-08-31 by EmperorArthur