kennetek/gridfinity-rebuilt-openscad #261: Feature Request/Idea: Square/Rectangular Holes

Shadoweclipse13, 2024-12-22:
Hello!!

Firstly, thank you for all the work you folks put into the generator. I use it a lot and it is fantastic!!

One of the ideas I have recently had (I will post the other separately) is to have square/rectangular holes, much like the cylinders (under the "bin style" option of the Gridfinity Rebuilt type), with the same/similar options: square/rectangular size (square would be even, maybe rectangular isn't necessary, just thinking out loud here), with side length (like cylinder diameter) and depth options.

phrxmd, 2024-12-22:
This is already possible with `cut_move`:

```
use <src/core/gridfinity-rebuilt-utility.scad>
use <src/core/gridfinity-rebuilt-holes.scad>

color("blue")
gridfinityInit(1, 1, height(6), 0, 42) {
    cut_move(x=.2, y=.2, w=1, h=1) {
          cube(size=[15,25,35], center=true);
    }
}
```

phrxmd, 2024-12-22:
This is already possible with `cut_move()`. Put this in a new file in the same folder as the `gridfinity-rebuilt-openscad` files:

```scad
use <src/core/gridfinity-rebuilt-utility.scad>
use <src/core/gridfinity-rebuilt-holes.scad>

color("blue"){
  gridfinityInit(1, 1, height(6), 0, 42) {
    cut_move(x=.2, y=.2, w=1, h=1) {
      cube(size=[15,25,35], center=true);
    };
  };
  gridfinityBase([1, 1]);
}
```

You might want to move beyond just playing around with the Customizer options, and start writing your own OpenSCAD code, that's the key to unlocking what the library can do.

Shadoweclipse13, 2024-12-28:
That fantastic that it's already a feature. Certainly no complaints there. I guess my point is just that not everyone understands coding, myself included. And adding some simple features, if it's not too difficult for you fine folks, would be awesome.

closed 2025-09-01 by EmperorArthur