kennetek/gridfinity-rebuilt-openscad #262: Feature Request/Idea: Offset/Asymmetric Divided Bins

Shadoweclipse13, 2024-12-22:
Hello again!

I had the idea that it could be useful to have asymmetrical divided bins: the divider closer to one side or the other, instead of just being centered. The option of being able to adjust the exact spacing from one side or the other could potentially be very handy for small/thin objects.

Thank you!!

phrxmd, 2024-12-22:
You can do this with the `cut` module, it supports fractional widths:

```
color("green") {
gridfinityInit(2, 1, height(6, gridz_define, style_lip, enable_zsnap), 0, sl=style_lip) {

    cut(0,0,.7,1);
    cut(.7,0,1.3,1);
}
gridfinityBase([2, 1], hole_options=hole_options, only_corners=only_corners, thumbscrew=enable_thumbscrew);
}
```

phrxmd, 2024-12-22:
You can do this with the `cut` module, it supports fractional widths. Put something like this in an SCAD file in the same directory as your `gridfinity-rebuilt-openscad` files:

```scad
use <src/core/gridfinity-rebuilt-utility.scad>
use <src/core/gridfinity-rebuilt-holes.scad>

color("green") {
  gridfinityInit(2, 1, height(6), 0, 42) {
    cut(0,0,.7,1);
    cut(.7,0,1.3,1);
  }
  gridfinityBase([2, 1], hole_options=bundle_hole_options(
    refined_hole=true, magnet_hole=false, screw_hole=false, 
    crush_ribs=true, chamfer=true, supportless=true), only_corners=true);
}
```

Try and move towards creating your own OpenSCAD code using the library, that will allow you to unlock much more functionality than if you just play around with the variables exposed to Customizer.

EmperorArthur, 2025-06-26:
Closing, as this is already possible via the customizer.

closed 2025-06-26 by EmperorArthur