kennetek/gridfinity-rebuilt-openscad #263: Scoop results in unnecessarily thick front wall

phrxmd, 2024-12-22:
For some reason as soon as I use subdivisions with a scoop, the front wall is much thicker than it needs to be. 
Consider this sample 1×1 bin:

```scad
use <src/core/gridfinity-rebuilt-utility.scad>
use <src/core/gridfinity-rebuilt-holes.scad>

$fa = 8;
$fs = 0.25;

color("green") {
  gridfinityInit(1, 1, height(6), 0, 42) {
    cutEqual(1, 1, place_tab=1, style_tab=0, scoop_weight = 1);
  }
  gridfinityBase([1, 1], hole_options=bundle_hole_options(
    refined_hole=true, magnet_hole=false, screw_hole=false, 
    crush_ribs=true, chamfer=true, supportless=true), only_corners=true);
}
```

As soon as `scoop_weight` is nonzero, the front wall is much thicker than the others:

[image]

The front 1×1 bin has `scoop_weight=1` , the rear 1×1 bin has `scoop_weight=0` (and thin walls), the 2×1 bin has `scoop_weight=0.5` (and thick walls).

Is there a bug in `cut()` or `cutEqual()` that leaves a thicker front wall as soon as a scoop is in place?

phrxmd, 2024-12-22:
I suspect this bit in `profile_cutter()` might be the culprit, as it's the only part where there is a check for nonzero `scoop`:

https://github.com/kennetek/gridfinity-rebuilt-openscad/blob/e1e5dcc49e8ffb5f8d718059830b4313d52f2ed7/src/core/gridfinity-rebuilt-utility.scad#L736-L740

But I don't understand the code well enough to begin experimenting with it.

phrxmd, 2024-12-22:
Ah, I get it now, the entire wall with the scoop is thicker to align with the front lip at the top, similar to what you get with Gridfinity Extended with the `fingerslide_lip_aligned` option. 

It would be nice if this was configurable.

adamsdv, 2025-04-14:
It looks like you have recognized why the wall gets thicker, it is so that the stacking-lip doesn't present a serious impediment to you pulling an item up the scoop and out of the bin.  It does NOT thicken when you have no lip (style lip = "remove lip subtractively" or "remove lit and retain height") so I don't see any reason to make it configurable.  

I'd suggest that you close the issue.

Thanks