kennetek/gridfinity-rebuilt-openscad #264: Feature Request: allow for scoop on both sides

muesli, 2024-12-27:
It would be nice to have an option to render the scoop on both the top & bottom side.

muesli, 2025-04-14:
@adamsdv Do you have this change pushed to a branch somewhere?

adamsdv, 2025-04-14:
I have just cleaned up the code and put it into the "double-scoop" branch of my repository (forked from the main one)
I plan on doing a PR after my currently pending one is addressed.
You can feel free to grab the double-scoop branch and see if you like the behavior.

adamsdv, 2025-04-21:
This feature is now in a pull request and you can see an animation of the effects of normal scoop, back scoop, both front and back and extreme values (>1.0) shown on a standard bin or one with tabs in pull-request #288 

If this is still of interest, you might increase the odds of it being accepted by a comment or two.