kennetek/gridfinity-rebuilt-openscad #276: How to use grid_dimensions?

gnestlinger, 2025-02-26:
After going through the commit history, I figured out that setting `grid_dimensions` is the way to create custom gridded bases for bins. 
So in `gridfinity-rebuilt-bins.scad` I added the parameter in the following way 
`gridfinityBase([gridx, gridy], grid_dimensions=[21,21], hole_options=hole_options, only_corners=only_corners, thumbscrew=enable_thumbscrew);`, 
aiming for a 21mm gridded base.

It somehow worked, but not really:
[image]

Is this the intended way or am I doing something wrong?

EmperorArthur, 2025-02-28:
Hi, maybe this will help you.

`gridfinityBase` is only for generating the base and not the bin.  That's why you're seeing the disconnect.

The total size of the base is [grid_dimension.x  * gridx, grid_dimension.y * gridy].  So if you cut the grid_dimension in half, you need to double gridx/ gridy.

gnestlinger, 2025-02-28:
Thanks for the reply! But I am sorry, this did not solve my problem.
Maybe I was not clear about what I want to achieve: I want to have the grid of the bins in a spacing of 21mm instead of the default 42mm.

adamsdv, 2025-04-12:
I have been using a much older version of Gridfinity Rebuilt.  I'm printed over a hundred bins of various types and sizes and almost immediately settled on 1/2 grids for my bases too.  So when I grabbed the new gridfinity repo, I was also bewildered by the removal of the div_base_x and div_base_y settings in the configurator.

I would like to re-implement those as options to 'force' the division that you want.  It would touch a couple files gridfinity-rebuilt-bins.scad and gridfinity-rebuilt-baseplate.scad.  Perhaps other people would like that available again in bins?

adamsdv, 2025-04-12:
I have pull request with these changes implemented.  We will see if it Is accepted into the repository.

Sperryfreak01, 2025-04-13:
What lucky timing, I just realized I needed this today.  I tried your branch and haven't figured out a way to have the x and y dimensions be different.

adamsdv, 2025-04-14:
For X and Y divisions to be different just pass a vector/list instead of a scalar.
For example [1,2] or [1,3]
rather than 2

adamsdv, 2025-04-14:
If you are using the customizer in OpenSCAD, with gridfinity-rebuilt-bins.scad open (from my branch) there is a section at the bottom named "Base Minimum Divisions" where div_base_x and div_base_y can be set independently.  so you can observe the results that way and see in the code how to pass the parameters.

PR #287 (2025-04-19): Simplify how half grid bins are created.
* Instead of trying to automatically calculate it in gridfinityBase, add a toggle and pass grid_dimensions everywhere.
* Ads support for half grid bins to gridfinitiy-rebuilt-lite.
* Fixes issue with gridfinity-rebuilt-lite passing a scalar instead of grid_dimensions.

* Fixes #281
* Fixes #276
* Supersedes #284
closed 2025-04-19 by EmperorArthur