kennetek/gridfinity-rebuilt-openscad #277: Internal fillet radius does not match spec

sbrocket, 2025-03-13:
https://github.com/kennetek/gridfinity-rebuilt-openscad/blob/993814227204b942fc1c13e64a604427f835742a/src/core/standard.scad#L66-L67

This radius seems to be incorrectly set, resulting in an inner radius for the stacking lip that does not match the outer radii. It should match the 1.6mm diameter shown here, i.e. `r_f2 = 0.8`.

[image]

Not at my computer at the moment but will attach photos with and without this radii tweaked later.

EmperorArthur, 2025-03-18:
The problem you're seeing is related to how the different parts interact.

The walls and lip are created, then the internals are filled, and finally the internals are hollowed out in the shape(s) defined by the user.

The fix would be to mandate walls as thick as the stacking lip, but that reduces available volume and takes more material.

Overall, I'd say actually just changing $r_f2 might be the best option.

EmperorArthur, 2025-03-23:
I've submitted a PR to move the internal fill down to the correct height.

It doesn't fix the radius issue, but will allow stacking.  Even with multiple compartments, or if the fill is solid.

[image]

PR #280 (2025-03-23): Fix automatic inernal_height.
Bin fill height did not take stacking into account.
Fixes #277
commit 3c4c1a00 (2025-03-23): Merge pull request #280 from EmperorArthur/itnernals

Fix automatic inernal_height.

Bin fill height did not take stacking into account.

Fixes #277
commit 7bcc3f5e (2025-03-23): Fix automatic inernal_height.

Bin fill height did not take stacking into account.
Fixes #277
closed 2025-03-23 by EmperorArthur