kennetek/gridfinity-rebuilt-openscad #291: Half-Grid change in PR #287 introduced bug in tab placement and linear divisions

adamsdv, 2025-04-27:
The half-grid change introduced in PR #287 has introduced a few bugs (probably related to l_grid & GRID_DIMENSIONS_MM being used in places rather than the newly passed parameter of 'grid_dimensions' introduced in gridfinityInit.

My guess is that some underlying modules and functions still use the globals rather than the passed parameter.

a couple examples are depicted below:

1) a 2x2x4 bin with divx = 1 and divy = 4, followed by the same sized bin with half-grid set (so 4x4x4) with divx = 1 and divy = 4.  so the exact same bin size with the exact bin divisions but with base grid being standard 42mm and being 21mm.

The results: 

[image]

[image]

2) a 3x2x4 bin divx=2, divy=2, tabs set to auto, everywhere. verses half-grid bin that is 6x4x4 with the same parameters identical

The results:

[image]

[image]

I am going to explore the core issue(s) if I have the time, but since I already have a PR pending and some other things 'In process' locally, I am hoping that either PR #288 is closed or someone else can find and fix this one.  My intuition says it may require small touches in a lot of modules and files to eliminate the dependence on the globals and rather use the parameter passed in gridfinityInit. 

If that's the case, I'm afraid that my proposed solution would run afoul of the maintainer's preference, as PR #284 did.

I'm looking for feedback on the vision and requirements of the project maintainers.  This is really, as far as I can determine, far and away the best Gridfinity library available and you guys have done so much great work. So I really want to help keep it top-notch and getting better every day.  

Thanks

adamsdv, 2025-04-27:
gridfinity-rebuilt-utility.scad
- cut_move_unsafe: uses l_grid and has no parameter for grid_dimensions
- block_cutter: uses l_grid and has no parameter for grid_dimensions
- cutCylinders:  uses l_grid and has no parameter for grid_dimensions
- gridfinityBase: uses GRID_DIMENSIONS_MM (it does have grid_dimensions passed as a parameter), the gap calculation MAY still be OK, I haven't checked it completely yet
- gridfinity_base_lite: same issue as gridfinityBase.

Changing the parameters to each of these to add grid_dimensions and eliminate the use of the globals, then to climb each one's call-tree to assure that this parameter is passed everywhere.  Then to check all the other files in the entire project for similar dependencies and use of the globals....

That just seems like a poor solution and more-wide ranging than I want to undertake right now.

Thoughts?

adamsdv, 2025-04-28:
More examination seems to make it clear that the widespread use in the project files, modules and functions of l_grid, GRID_DIMENSIONS_MM, BASE_TOP_DIMENSIONS and other constants derived from these values make passing a modified grid_dimensions as a parameter is not very clean, as it will take too much fiddling with other functions and modules throughout to make it all work correctly.  I have kind of gotten it working locally, but it is crazy intertwined and I am not at all sure that I have actually got it working correctly in all cases, certainly not even testing lite and vase-mode things. There is no way I would commit these changes.  This half_grid change introduced in PR #287 has really broken quite a lot of stuff with half-grid sized bins.

Is it possible to just roll back that change?

@EmperorArthur, I appreciate that you said you don't really like half-bin steps.  It seems you are currently one of the most active admins here, so I understand.  I disagree, but I defer to your position.  If it is not going to work correctly, I think we need to roll back PR #287.  Is there any chance of that?

pferriola, 2025-06-02:
I agree that #287 should probably be rolled back if it's not going to be fixed soon.  I recently updated to latest and it broke a lot of things for me and it would take quite a bit to untangle it.  I spent a little time on it and it's going to strand me on an old revision if we can't get this back to where it works properly.

EmperorArthur, 2025-06-07:
@adamsdv and @pferriola 

Sorry, I've been absent for so long.  I've been hard at work on a V2 of Gridfinity bins.  It's not fully done, but using that version the problem is fixed.  #305

@adamsdv To clarify, half-size has a ton of "gotchas" in relation to spacing. For example, if we scaled everything then bins would catch when being used with regular spacing.  Then there's magnet holes overlapping. Lots of work to deviate from the standard, but I get the utility.

closed 2025-08-31 by EmperorArthur