kennetek/gridfinity-rebuilt-openscad #304: Documentation is out of date

reinux, 2025-05-29:
`gridfinityBase` currently has a signature of: 

```openscad
module gridfinityBase(grid_size, grid_dimensions=GRID_DIMENSIONS_MM, hole_options=bundle_hole_options(), off=0, final_cut=true, only_corners=false, thumbscrew=false)
```

But is described in the docs as being of the form:

```openscad
gridfinityBase (gridx, gridy, length, div_base_x, div_base_y, style_hole)
```

Could you please let us know which commit is in line with the docs?

Thanks!

EmperorArthur, 2025-06-26:
My fault.  The docs are all horribly outdated.

If you want to go back to that version, you're going to have to go back to over a year ago.  Commit 2e0e9bd3632841574a26e3360daccb9ae9d41f29 was the one prior to the change and is from 2024-04-21.  Even that has an extra argument!  If you want to match the docs exactly you're going to have to go to afce2f90ec74e6a4a14af2fc0a022a5b9904ccba from 2022-10-10.

I will warn you.  Going back that far you will have some issues, including the bin parameters not quite matching the spec.

EmperorArthur, 2025-09-01:
V2.0.0 drastically changed the library.  Making the documentation even more outdated.  I am currently looking for the best OpenSCAD documentation generator.  Most functions are now documented using doxygen syntax, and could be used to auto-generate some of the docs.