kennetek/gridfinity-rebuilt-openscad #311: ERROR: Parser error: syntax error in file gridfinity-rebuilt-bins.scad, line 125

cskoven, 2025-09-13:
I just cloned the repo (git clone ...) and tried to open/render the "gridfinity-rebuilt-bins.scad" on my newly installed computer (Debian GNU/Linux 13 (trixie), OpenSCAD version 2021.01).

But I immediately get the following:

> [ERROR: Parser error: syntax error in file ../../gridfinity-rebuilt-bins.scad, line 125](https://github.com/kennetek/gridfinity-rebuilt-openscad/issues/125,/home/cskoven/git/gridfinity-rebuilt-openscad/gridfinity-rebuilt-bins.scad) 
> Execution aborted

The baseplate.scad opens and renders fine. What am I missing?

cskoven, 2025-09-13:
I tried installing the "nightly build" of openscad-nightly (openscad-nightly; 2025.09.11.nightly (git d38a625)). Now it runs fine.
The README states under "Recommendations" that: "It is not a requirement to use development versions of OpenSCAD.". That is not my experience.

romosborne, 2025-09-28:
Looks like it's just an extra trailing comma - I've opened a pull request to fix the issue.

If you want to just run it locally, remove the comma on line 124